blackmagic-debug/blackmagic #2018: [Cortex M55] UsageFault when trying to manipulate execution flow

florg-32, 2024-12-13:
Hey, when trying to `step`, `next` or once a set breakpoint is hit, the debugger blocks indefinitely and when halted manually (Ctrl+C) I can see that the CPU landed in the fault handler, with UNDEFINSTR set in the UsageFault register.

Just halting with Ctrl+C and `continue` work fine though.

The MCU I am working with has 2 Cortex M55, both revision r1p0, with 
- VFPv5
- MVE
- Armv8.1-M Main Extension (16-bit and 32-bit Thumb® instruction set)
- Armv8-M Security Extension (TrustZone)
- DSP Extension
- DSP Debug Extension
- Reliability Availability and Serviceability (RAS) Extension
- Unprivileged Debug Extension
- MPU
- BPU, DWT, PMU, Coresight-compliant DAP

---

<details>
<summary>❯ ./build/blackmagic -tv 7</summary>

```
Black Magic Debug App v1.10.0-1550-gdeeece3a
 for Black Magic Probe, ST-Link v2 and v3, CMSIS-DAP, J-Link and FTDI (MPSSE)
Using 1d50:6018 81D749A3 Black Magic Debug
 Black Magic Probe v1.10.0-1550-gdeeece3a
Running in Test Mode
Target voltage: 1.8V
Speed set to 1.951MHz for SWD
Switching from JTAG to dormant
Switching out of dormant state into SWD
Handling SWD multi-drop, TARGETID 0x07570257
DP DPIDR 0x4c013477 (v3 MINDP rev4) designer 0x43b partno 0xc0
TARGETID 0x07570257 designer 0x22b partno 0x7570
DP DPIDR1 0x000000a0 32-bit addressing
0 0x000000000: Debug component - CoreSight ROM (ROM Table) (PIDR = 0x02000ab758 DEVTYPE = 0x00 ARCHID = 0x0af7)
ROM Table: BASE=0x000000000 SYSMEM=0, Manufacturer 22b Partno 758 (PIDR = 0x02000ab758)
 0 0x000010000: Debug component - Unknown (PIDR = 0x04000bb9f0 DEVTYPE = 0x00 ARCHID = 0x0000)
 1 0x000020000: Debug component - Unknown (PIDR = 0x04000bb9ef DEVTYPE = 0x00 ARCHID = 0x0a57)
2 0x000030000: Debug component - CoreSight ROM (ROM Table) (PIDR = 0x02000ab759 DEVTYPE = 0x00 ARCHID = 0x0af7)
ROM Table: BASE=0x000030000 SYSMEM=0, Manufacturer 22b Partno 759 (PIDR = 0x02000ab759)
0 0x000040000: Debug component - ADIv6 MEM-APv2 (Memory Access Port) (PIDR = 0x04003bb9e3 DEVTYPE = 0x00 ARCHID = 0x0a17)
AP   0: IDR=34770008 -> Disabled
1 0x000050000: Debug component - ADIv6 MEM-APv2 (Memory Access Port) (PIDR = 0x04002bb9e2 DEVTYPE = 0x00 ARCHID = 0x0a17)
AP   1: IDR=24770006 CFG=000101a0 BASE=00000000 CSW=90800040 (APB4/5-AP var0 rev2)
0 0x000000000: 0x02000ab75a Non-ARM component ignored
  -> designer: 22b, part no: 75a
2 0x000060000: Debug component - CoreSight ROM (ROM Table) (PIDR = 0x02000ab75b DEVTYPE = 0x00 ARCHID = 0x0af7)
ROM Table: BASE=0x000060000 SYSMEM=0, Manufacturer 22b Partno 75b (PIDR = 0x02000ab75b)
0 0x000070000: Debug component - CoreSight MTE (Memory Tagging Extensioon) (PIDR = 0x04003bb9e4 DEVTYPE = 0x00 ARCHID = 0x0a17)
ROM Table: END
3 0x000100000: Debug component - CoreSight ATBF (ATB Funnel) (PIDR = 0x04002bb9eb DEVTYPE = 0x12 ARCHID = 0x0000)
4 0x000110000: Debug component - CoreSight ATBR (ATB Replicator) (PIDR = 0x04002bb9ec DEVTYPE = 0x22 ARCHID = 0x0000)
5 0x000120000: Debug component - CoreSight TPIU (Trace Port Interface Unit) (PIDR = 0x04002bb9e7 DEVTYPE = 0x11 ARCHID = 0x0000)
6 0x000130000: Debug component - CoreSight CTI (Cross Trigger Interface) (PIDR = 0x04003bb9ed DEVTYPE = 0x14 ARCHID = 0x1a14)
7 0x000140000: Debug component - CoreSight TCM (Trace Memory Controller) (PIDR = 0x04005bb9e8 DEVTYPE = 0x21 ARCHID = 0x0000)
8 0x000150000: Debug component - CoreSight CATU (CoreSight Address Translation Unit) (PIDR = 0x04001bb9ee DEVTYPE = 0x00 ARCHID = 0x0000)
9 0x000160000: Debug component - CoreSight CTI (Cross Trigger Interface) (PIDR = 0x04003bb9ed DEVTYPE = 0x14 ARCHID = 0x1a14)
10 0x000200000: Debug component - ADIv6 MEM-APv2 (Memory Access Port) (PIDR = 0x04003bb9e3 DEVTYPE = 0x00 ARCHID = 0x0a17)
AP  10: IDR=34770008 CFG=000101a0 BASE=e00ff000 CSW=a3800040 (AHB5-AP var0 rev3)
ROM Table: BASE=0xe00ff000 SYSMEM=1, Manufacturer 43b Partno 4d2 (PIDR = 0x04000bb4d2)
 0 0x0e000e000: Debug component - Cortex-M55 (Ssystem Control Space) (PIDR = 0x04000bbd22 DEVTYPE = 0x00 ARCHID = 0x2a04)
 -> cortexm_probe
CPUID 0x411fd220 (M55 var 1 rev 0)
cortexm_probe: Examining Part ID 0x7570, AP Part ID: 0x0759
Please report unknown device with Designer 0x22b Part ID 0x7570
 1 0x0e0001000: Debug component - Cortex-M55 (Data Watchpoint and Trace) (PIDR = 0x04000bbd22 DEVTYPE = 0x00 ARCHID = 0x1a02)
 2 0x0e0002000: Debug component - Cortex-M55 (Breakpoint Unit) (PIDR = 0x04000bbd22 DEVTYPE = 0x00 ARCHID = 0x1a03)
 3 0x0e0000000: Debug component - Cortex-M55 (Instrumentation Trace Macrocell) (PIDR = 0x04000bbd22 DEVTYPE = 0x43 ARCHID = 0x1a01)
 4 Entry 0xfff41002 -> Not present
 5 0x0e0041000: Debug component - Cortex-M55 (Embedded Trace) (PIDR = 0x04003bbd22 DEVTYPE = 0x13 ARCHID = 0x4a13)
 6 0x0e0003000: Debug component - Cortex-M55 (Performance Monitoring Unit) (PIDR = 0x04000bbd22 DEVTYPE = 0x16 ARCHID = 0x0a06)
 7 0x0e0042000: Debug component - Cortex-M55 (Cross Trigger) (PIDR = 0x04000bbd22 DEVTYPE = 0x14 ARCHID = 0x1a14)
 8 Entry 0xfff47002 -> Not present
 ROM Table: END
Resuming target
11 0x000300000: Debug component - ADIv6 MEM-APv2 (Memory Access Port) (PIDR = 0x04003bb9e3 DEVTYPE = 0x00 ARCHID = 0x0a17)
AP  11: IDR=34770008 CFG=000101a0 BASE=e00ff000 CSW=a3800040 (AHB5-AP var0 rev3)
ROM Table: BASE=0xe00ff000 SYSMEM=1, Manufacturer 43b Partno 4d2 (PIDR = 0x04000bb4d2)
 0 0x0e000e000: Debug component - Cortex-M55 (Ssystem Control Space) (PIDR = 0x04000bbd22 DEVTYPE = 0x00 ARCHID = 0x2a04)
 -> cortexm_probe
CPUID 0x411fd220 (M55 var 1 rev 0)
cortexm_probe: Examining Part ID 0x7570, AP Part ID: 0x0759
Please report unknown device with Designer 0x22b Part ID 0x7570
 1 0x0e0001000: Debug component - Cortex-M55 (Data Watchpoint and Trace) (PIDR = 0x04000bbd22 DEVTYPE = 0x00 ARCHID = 0x1a02)
 2 0x0e0002000: Debug component - Cortex-M55 (Breakpoint Unit) (PIDR = 0x04000bbd22 DEVTYPE = 0x00 ARCHID = 0x1a03)
 3 0x0e0000000: Debug component - Cortex-M55 (Instrumentation Trace Macrocell) (PIDR = 0x04000bbd22 DEVTYPE = 0x43 ARCHID = 0x1a01)
 4 Entry 0xfff41002 -> Not present
 5 0x0e0041000: Debug component - Cortex-M55 (Embedded Trace) (PIDR = 0x04003bbd22 DEVTYPE = 0x13 ARCHID = 0x4a13)
 6 0x0e0003000: Debug component - Cortex-M55 (Performance Monitoring Unit) (PIDR = 0x04000bbd22 DEVTYPE = 0x16 ARCHID = 0x0a06)
 7 0x0e0042000: Debug component - Cortex-M55 (Cross Trigger) (PIDR = 0x04000bbd22 DEVTYPE = 0x14 ARCHID = 0x1a14)
 8 Entry 0xfff47002 -> Not present
 ROM Table: END
Resuming target
ROM Table: END
ROM Table: END
***  1   Unknown ARM Cortex-M Designer 0x22b Part ID 0x7570 M55
***  2   Unknown ARM Cortex-M Designer 0x22b Part ID 0x7570 M55
Attaching to target..
Halting target
Target halted: Halt requested
Attach success
Detaching from target
```

</details>

dragonmux, 2024-12-14:
> I do not know if it is related but also after attaching to one of the cores, I see two threads and inferiors while the second core is held in reset.

What version of GDB is this with as that seems extremely wrong (it should be presenting one thread always, and there should only be the one inferior). It is possible that it's related to the primary issue seen.

florg-32, 2024-12-16:
I'm using 

```
❯ gdb-multiarch --version
GNU gdb (Ubuntu 12.1-0ubuntu1~22.04.2) 12.1
```

dragonmux, 2024-12-16:
We're glad you've been able to figure it out! Huzzah!