blackmagic-debug/blackmagic #2033: STM32WLxx: jtag_scan reports unknown device while swd_scan correctly identifies device

AJ528, 2024-12-21:
Hello,

I recently got a black magic probe and have been attempting to program an STM32WL55CC. I have the full JTAG interface brought out to a standard ARM 10 pin debug header, but the  black magic probe cannot seem to correctly identify the MCU when using jtag_scan. Even weirder, jtag_scan outputs a different value depending on if swd_scan was run before it.

Black Magic Probe is running v1.10.2, and I am using gdb-multiarch version 12.1.

My System Info:
```
  Kernel: 5.15.0-128-generic x86_64 bits: 64 compiler: gcc v: 11.4.0 Desktop: Cinnamon 6.0.4
    tk: GTK 3.24.33 wm: muffin vt: 7 dm: LightDM 1.30.0 Distro: Linux Mint 21.3 Virginia
    base: Ubuntu 22.04 jammy
```

GDB log showing the issue:
```
GNU gdb (Ubuntu 12.1-0ubuntu1~22.04.2) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb) target extended-remote /dev/ttyACM0
Remote debugging using /dev/ttyACM0
(gdb) monitor version
Black Magic Probe v1.10.2, Hardware Version 6
Copyright (C) 2010-2023 Black Magic Debug Project
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>

(gdb) monitor jtag_scan
Target voltage: 3.3V
Please report unknown device with Designer 0x20 Part ID 0x497
Available Targets:
No. Att Driver
*** 1   Unknown ARM Cortex-M Designer 0x20 Part ID 0x497 M4
(gdb) monitor swd_scan
Target voltage: 3.3V
Available Targets:
No. Att Driver
 1      STM32WLxx M4
 2      STM32WLxx M0+
(gdb) monitor jtag_scan
Target voltage: 3.3V
Please report unknown device with Designer 0x20 Part ID 0x497
Please report unknown device with Designer 0x20 Part ID 0x497
Available Targets:
No. Att Driver
*** 1   Unknown ARM Cortex-M Designer 0x20 Part ID 0x497 M4
*** 2   Unknown ARM Cortex-M Designer 0x20 Part ID 0x497 M0+
(gdb) q
```

dragonmux, 2024-12-22:
👋🏼 Thank you for the report. Please try upgrading your firmware to `main` which is currently undergoing final stabilisation for v2.0-rc0's release. This issue should have been fixed in #1952

AJ528, 2024-12-22:
Hello,
Thank you for the quick reply! I updated my firmware to `main`, and that appears to be a regression. Neither `swd_scan` nor `jtag_scan` correctly identify my device now. I had a little trouble building the firmware to fit inside the ROM, but I used the `native-remote.ini` crossfile, set my targets to `cortexm` and `stm`, and disabled RTT.

Here is the new log:
```
GNU gdb (Ubuntu 12.1-0ubuntu1~22.04.2) 12.1
(gdb) target extended-remote /dev/ttyACM0
Remote debugging using /dev/ttyACM0
(gdb) monitor version
Black Magic Probe v1.10.0-1591-gc37a38e4, Hardware Version 6
Copyright (C) 2010-2024 Black Magic Debug Project
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>

(gdb) monitor jtag_scan
Target voltage: 3.3V
Please report unknown device with Designer 0x20 Part ID 0x497
Available Targets:
No. Att Driver
*** 1   Unknown ARM Cortex-M Designer 0x20 Part ID 0x497 M4
(gdb) monitor swd_scan
Target voltage: 3.3V
Please report unknown device with Designer 0x20 Part ID 0x4970
Available Targets:
No. Att Driver
*** 1   Unknown ARM Cortex-M Designer 0x20 Part ID 0x4970 M4
(gdb) monitor jtag_scan
Target voltage: 3.3V
Please report unknown device with Designer 0x20 Part ID 0x497
Available Targets:
No. Att Driver
*** 1   Unknown ARM Cortex-M Designer 0x20 Part ID 0x497 M4
(gdb) q
```

dragonmux, 2024-12-22:
What you are seeing is normal behaviour for a remote-only build of BMD. Please use one of the [nightly builds](https://nightly.link/blackmagic-debug/blackmagic/workflows/build-and-upload/main) of the firmware or use at least GCC 12.2.Rel1 from ARM to compile the firmware.

For your use case, you need either the [STM-and-clones](https://github.com/blackmagic-debug/blackmagic/blob/main/cross-file/native-st-clones.ini) specific configuration or the [default baseline](https://github.com/blackmagic-debug/blackmagic/blob/main/cross-file/native.ini) configuration for native. It will fit when built this way with a new enough compiler.

The remote-only configuration you have used turns off all target support, and is designed and intended for use with BMDA to drive the probe. It contains only the remote protocol and acceleration systems and core firmware components to have a successful build. It does not contain any target support whatsoever.

AJ528, 2024-12-22:
Oh, okay that's good to know. I didn't realize the intended use-case for the remote-only configuration. I will work on updating my GCC to compile the latest default baseline from `main`. In the meantime, I downloaded the nightly build and flashed `blackmagic_native_firmware.elf` to my BMP. Rerunning my test on that firmware also appears broken. After running `monitor swd_scan`, the probe just prints `ignoring packet error` ever 5-10 seconds until it hangs and I have to kill it.

GDB log:
```
GNU gdb (Ubuntu 12.1-0ubuntu1~22.04.2) 12.1
(gdb) target extended-remote /dev/ttyACM0
Remote debugging using /dev/ttyACM0
(gdb) monitor version
Black Magic Probe 804da84, Hardware Version 6
Copyright (C) 2010-2024 Black Magic Debug Project
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>

(gdb) monitor swd_scan
Target voltage: 3.3V
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
^C
^C
^C^C^C^C^C
Remote connection closed
(gdb) Quit
(gdb) 
"monitor" command not supported by this target.
(gdb) q
```

dragonmux, 2024-12-23:
That's deeply disconcerting behaviour as that indicates something when wrong during scan and the probe crashed/hung. Do you see the same for JTAG scans too?

AJ528, 2024-12-23:
yes, same behavior with `jtag_scan`. I am happy to run any other diagnostic tests that may help narrow down the issue.

dragonmux, 2024-12-23:
That's disconcerting. Could you please try using BMDA (available for your platform from the nightlies linked before) and running it as `blackmagic-bmda -tv 5`? This will do a SWD scan of the target and report all the ROM tables and other information discovered, try and attach to the first found target, and then display a summary of the Flash and RAM mappings and exit. It should allow us to see what's going on with discovery of that target.

AJ528, 2024-12-23:
sure thing. The output from `blackmagic-bmda -tv 5` is below. It seems like it ran just fine. I did verify the problem is still occurring though.

```
Black Magic Debug App 804da84
 for Black Magic Probe, ST-Link v2 and v3, CMSIS-DAP, J-Link and FTDI (MPSSE)
Using 1d50:6018 81D264A5 Black Magic Debug
 Black Magic Probe 804da84
Running in Test Mode
Target voltage: 3.3V
Speed set to 1.951MHz for SWD
Switching from JTAG to dormant
Switching out of dormant state into SWD
Handling SWD multi-drop, TARGETID 0x04970041
DP DPIDR 0x6ba02477 (v2 rev6) designer 0x43b partno 0xba
TARGETID 0x04970041 designer 0x20 partno 0x4970
AP   0: IDR=24770011 CFG=00000000 BASE=e00ff000 CSW=e3000040 (AHB3-AP var1 rev2)
Halt via DHCSR(00030003): success after 11ms
ROM Table: BASE=0xe00ff000 SYSMEM=1, Manufacturer 020 Partno 497 (PIDR = 0x00000a0497)
0 0x0e000e000: Generic IP component - Cortex-M3 SCS (System Control Space) (PIDR = 0x04000bb000 DEVTYPE = 0x00 ARCHID = 0x0000)
-> cortexm_probe
CPUID 0x410fc241 (M4 var 0 rev 1)
cortexm_probe: Examining Part ID 0x4970, AP Part ID: 0x0497
Calling stm32f1_probe
Calling stm32f4_probe
Calling stm32h5_probe
Calling stm32h7_probe
Calling stm32mp15_cm4_probe
Calling stm32l0_probe
Calling stm32l1_probe
Calling stm32l4_probe
1 0x0e0001000: Generic IP component - Cortex-M3 DWT (Data Watchpoint and Trace) (PIDR = 0x04003bb002 DEVTYPE = 0x00 ARCHID = 0x0000)
2 0x0e0002000: Generic IP component - Cortex-M3 FBP (Flash Patch and Breakpoint) (PIDR = 0x04002bb003 DEVTYPE = 0x00 ARCHID = 0x0000)
3 0x0e0000000: Generic IP component - Cortex-M3 ITM (Instrumentation Trace Module) (PIDR = 0x04003bb001 DEVTYPE = 0x00 ARCHID = 0x0000)
4 0x0e0040000: Debug component - Cortex-M4 TPIU (Trace Port Interface Unit) (PIDR = 0x04000bb9a1 DEVTYPE = 0x11 ARCHID = 0x0000)
5 Entry 0xfff42002 -> Not present
6 0x0e0043000: Debug component - CoreSight CTI (Cross Trigger) (PIDR = 0x04005bb906 DEVTYPE = 0x14 ARCHID = 0x0000)
ROM Table: END
Resuming target
AP   1: IDR=84770001 CFG=00000000 BASE=f0000000 CSW=a3800040 (AHB3-AP var0 rev8)
Patching Designer code 420 -> 020
ROM Table: BASE=0xf0000000 SYSMEM=1, Manufacturer 020 Partno 497 (PIDR = 0x04000a0497)
ROM Table: BASE=0xe00ff000 SYSMEM=1, Manufacturer 43b Partno 4c0 (PIDR = 0x04000bb4c0)
 0 0x0e000e000: Generic IP component - Cortex-M0 SCS (System Control Space) (PIDR = 0x04000bb008 DEVTYPE = 0x00 ARCHID = 0x0000)
 -> cortexm_probe
CPUID 0x410cc601 (M0+ var 0 rev 1)
cortexm_probe: Examining Part ID 0x4970, AP Part ID: 0x0497
Calling stm32f1_probe
Calling stm32f4_probe
Calling stm32h5_probe
Calling stm32h7_probe
Calling stm32mp15_cm4_probe
Calling stm32l0_probe
Calling stm32l1_probe
Calling stm32l4_probe
 1 0x0e0001000: Generic IP component - Cortex-M0 DWT (Data Watchpoint and Trace) (PIDR = 0x04000bb00a DEVTYPE = 0x00 ARCHID = 0x0000)
 2 0x0e0002000: Generic IP component - Cortex-M0 BPU (Breakpoint Unit) (PIDR = 0x04000bb00b DEVTYPE = 0x00 ARCHID = 0x0000)
 ROM Table: END
1 0x0f0001000: Debug component - Cortex-M0+ CTI (Cross Trigger Interface) (PIDR = 0x04000bb9a6 DEVTYPE = 0x14 ARCHID = 0x1a14)
2 Entry 0x00002002 -> Not present
3 Entry 0x10000002 -> Not present
ROM Table: END
Resuming target
***  1   STM32WLxx M4
***  2   STM32WLxx M0+
Attaching to target..
Halting target
Target halted: Halt requested
Attach success
RAM   Start: 0x10000000 length = 0x1f65000
RAM   Start: 0x20000000 length = 0x10000
Flash Start: 0x08000000 length = 0x40000 blocksize 0x800
Detaching from target
```

AJ528, 2024-12-23:
I also ran `blackmagic-bmda -v 5` in one window, then connected to the server using `target extended-remote :2000`. It seems to work in this configuration. Logs are below.

GDB Session:
```
(gdb) target extended-remote :2000
Remote debugging using :2000
(gdb) monitor version
Black Magic Debug App 804da84
Using a Black Magic Probe (Black Magic Debug), 804da84
Copyright (C) 2010-2024 Black Magic Debug Project
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>

(gdb) monitor swd_scan
Target voltage: 3.3V
Available Targets:
No. Att Driver
 1      STM32WLxx M4
 2      STM32WLxx M0+
(gdb) q
```

BMDA Log:
```
./blackmagic-bmda -v 5
Black Magic Debug App 804da84
 for Black Magic Probe, ST-Link v2 and v3, CMSIS-DAP, J-Link and FTDI (MPSSE)
Using 1d50:6018 81D264A5 Black Magic Debug
 Black Magic Probe 804da84
Setting V6ONLY to off for dual stack listening.
Listening on TCP port: 2000
Got connection
Speed set to 1.951MHz for SWD
Switching from JTAG to dormant
Switching out of dormant state into SWD
Handling SWD multi-drop, TARGETID 0x04970041
DP DPIDR 0x6ba02477 (v2 rev6) designer 0x43b partno 0xba
TARGETID 0x04970041 designer 0x20 partno 0x4970
AP   0: IDR=24770011 CFG=00000000 BASE=e00ff000 CSW=e3000040 (AHB3-AP var1 rev2)
Halt via DHCSR(01030003): success after 7ms
ROM Table: BASE=0xe00ff000 SYSMEM=1, Manufacturer 020 Partno 497 (PIDR = 0x00000a0497)
0 0x0e000e000: Generic IP component - Cortex-M3 SCS (System Control Space) (PIDR = 0x04000bb000 DEVTYPE = 0x00 ARCHID = 0x0000)
-> cortexm_probe
CPUID 0x410fc241 (M4 var 0 rev 1)
cortexm_probe: Examining Part ID 0x4970, AP Part ID: 0x0497
Calling stm32f1_probe
Calling stm32f4_probe
Calling stm32h5_probe
Calling stm32h7_probe
Calling stm32mp15_cm4_probe
Calling stm32l0_probe
Calling stm32l1_probe
Calling stm32l4_probe
1 0x0e0001000: Generic IP component - Cortex-M3 DWT (Data Watchpoint and Trace) (PIDR = 0x04003bb002 DEVTYPE = 0x00 ARCHID = 0x0000)
2 0x0e0002000: Generic IP component - Cortex-M3 FBP (Flash Patch and Breakpoint) (PIDR = 0x04002bb003 DEVTYPE = 0x00 ARCHID = 0x0000)
3 0x0e0000000: Generic IP component - Cortex-M3 ITM (Instrumentation Trace Module) (PIDR = 0x04003bb001 DEVTYPE = 0x00 ARCHID = 0x0000)
4 0x0e0040000: Debug component - Cortex-M4 TPIU (Trace Port Interface Unit) (PIDR = 0x04000bb9a1 DEVTYPE = 0x11 ARCHID = 0x0000)
5 Entry 0xfff42002 -> Not present
6 0x0e0043000: Debug component - CoreSight CTI (Cross Trigger) (PIDR = 0x04005bb906 DEVTYPE = 0x14 ARCHID = 0x0000)
ROM Table: END
Resuming target
AP   1: IDR=84770001 CFG=00000000 BASE=f0000000 CSW=a3800040 (AHB3-AP var0 rev8)
Patching Designer code 420 -> 020
ROM Table: BASE=0xf0000000 SYSMEM=1, Manufacturer 020 Partno 497 (PIDR = 0x04000a0497)
ROM Table: BASE=0xe00ff000 SYSMEM=1, Manufacturer 43b Partno 4c0 (PIDR = 0x04000bb4c0)
 0 0x0e000e000: Generic IP component - Cortex-M0 SCS (System Control Space) (PIDR = 0x04000bb008 DEVTYPE = 0x00 ARCHID = 0x0000)
 -> cortexm_probe
CPUID 0x410cc601 (M0+ var 0 rev 1)
cortexm_probe: Examining Part ID 0x4970, AP Part ID: 0x0497
Calling stm32f1_probe
Calling stm32f4_probe
Calling stm32h5_probe
Calling stm32h7_probe
Calling stm32mp15_cm4_probe
Calling stm32l0_probe
Calling stm32l1_probe
Calling stm32l4_probe
 1 0x0e0001000: Generic IP component - Cortex-M0 DWT (Data Watchpoint and Trace) (PIDR = 0x04000bb00a DEVTYPE = 0x00 ARCHID = 0x0000)
 2 0x0e0002000: Generic IP component - Cortex-M0 BPU (Breakpoint Unit) (PIDR = 0x04000bb00b DEVTYPE = 0x00 ARCHID = 0x0000)
 ROM Table: END
1 0x0f0001000: Debug component - Cortex-M0+ CTI (Cross Trigger Interface) (PIDR = 0x04000bb9a6 DEVTYPE = 0x14 ARCHID = 0x1a14)
2 Entry 0x00002002 -> Not present
3 Entry 0x10000002 -> Not present
ROM Table: END
Resuming target
Error on socket 9, got error 11: Resource temporarily unavailable
```

AJ528, 2024-12-24:
I had an idea and tried running `swd_scan`without anything connected. That worked exactly as expected and just said "nothing is connected". So the issue occurs sometime after verifying a device is connected. Unfortunately, I'm not sure this helps narrow it down much.

dragonmux, 2024-12-24:
You are correct that it doesn't narrow it down at all. That BMDA works and as that runs the exact same code as on the probe for the actual target detection and discovery logic, it suggests one of two things: Either the probe is crashing due to something like an out-of-memory condition; there's some kind of bug in stm32l4.c which results in a bad pointer dereference which crashes the probe (but this feels improbable)

Best way to find out what's going on is if you have a second probe to be able to debug the first with - the BMP v2.3 board has a 4 pin SWD interface exposed down one edge on 1.27mm pitch header holes, this can be used to do Black Magic Inception (debugging the debugger from another) and if this is a crash of some kind, you'll be able to either catch the Hard Fault handler firing that way, or otherwise Ctrl + C execution to catch where it's become stuck.

AJ528, 2024-12-24:
Hmmm, I don't have a second BMP, but I do have an STM32 Nucleo board. I should be able to use the debugger on that to debug the BMP, yeah?

I have the BMP v2.3c. it looks like on my board the debug port has 5 pins. Do you know what the "TRA" pin is for? I know what the others are.

Also, it looks like I can easily pop a hole in the case for access to these pins. Is there a recommended pin header/connector combo that easily fits through the hole in the case?

dragonmux, 2024-12-24:
> Hmmm, I don't have a second BMP, but I do have an STM32 Nucleo board. I should be able to use the debugger on that to debug the BMP, yeah?

Yeah, if it's an ST-Link v2(.1) then you can use [stlink-tool](https://github.com/blackmagic-debug/stlink-tool) to re-Flash it over to the project firwmare (blackmagic_stlink_firmware.elf from the nightly firmware archive) and use that. It talks SWD only but that matches BMP's processor debug interface.

> I have the BMP v2.3c. it looks like on my board the debug port has 5 pins. Do you know what the "TRA" pin is for? I know what the others are.

The 5 pins on the debug connector are DIO -> SWDIO, GND, CLK -> SWCLK, TRA -> TRACESWO, and ~RST. SWO from the processor is presently unused however, so you can safely ignore this pin.

> Also, it looks like I can easily pop a hole in the case for access to these pins. Is there a recommended pin header/connector combo that easily fits through the hole in the case?

Standard 1.27mm pitch header will fit just fine, and any standard width 1.27mm socket should likewise fit great. You are correct that the case is designed to be able to be used with that header exposed in the back of the case for debugging the board.

[image]

AJ528, 2024-12-24:
I have an NUCLEO-WL55JC, which says it comes with an STLINK-V3. Should that work?

Also, what voltage does the debug interface use? 3.3V?

dragonmux, 2024-12-25:
> I have an NUCLEO-WL55JC, which says it comes with an STLINK-V3. Should that work?

You'll need to use BMDA with it, the chances you can load the stlinkv3 platform firmware on that are very slim.. BMDA from `main` should work with it though. You'll need to grab the serial number for it (either from `dmesg`, or by running `blackmagic-bmda -l` with it connected) and use that to have BMDA specifically connect to it via `blackmagic-bmda -s <serial>`.

The reason for saying use BMDA in this case is so you get use of the vector catch system for Cortex-M parts which is instrumental in debugging the firmware in this instance. It allows you to catch faults as they occur and is enabled out the box for reset and hard faults.

> Also, what voltage does the debug interface use? 3.3V?

Yes, correct - 3.3V.

AJ528, 2024-12-29:
Small update while I am waiting for debug cables to capture the crash  information. I downloaded `arm-none-eabi-gdb` v15.2, and the crash happens on that version of gdb as well. Log below:

```
~/Downloads/arm-gnu-toolchain-14.2.rel1-x86_64-arm-none-eabi/bin$ ./arm-none-eabi-gdb --version
GNU gdb (Arm GNU Toolchain 14.2.Rel1 (Build arm-14.52)) 15.2.90.20241130-git
Copyright (C) 2024 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
~/Downloads/arm-gnu-toolchain-14.2.rel1-x86_64-arm-none-eabi/bin$ ./arm-none-eabi-gdb
(gdb) target extended-remote /dev/ttyACM0
Remote debugging using /dev/ttyACM0
(gdb) monitor version
Black Magic Probe 804da84, Hardware Version 6
Copyright (C) 2010-2024 Black Magic Debug Project
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>

(gdb) monitor swd_scan
Target voltage: 3.3V
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
^C
^C
^C
^C^C^C^C^C^C^C
^C
^C
Remote connection closed
(gdb) Quit
(gdb) 
"monitor" command not supported by this target.
(gdb) q
```

dragonmux, 2025-01-07:
Okay, so - there seems to be a few things in play here, and part of the bug should have been fixed by ALTracer's #2040 which addressed two bugs in the target support for this.

The first thing we hit is that, if scanning without connecting under reset - at least on the firmware that comes on the Nucleo - then the chip will go no-contact after the first AP is scanned, when the CPU is allowed to unhalt and before we can scan AP1. This results in one flavour of crash, but is ultimately unavoidable. This needs some extra handling built around `adiv5_new_ap()` to allow the operation to fail gracefully.

The second issue we hit is that the remote protocol was not properly conveying to the adaptor what the target ID needed is for protocol recovery to run properly and re-select the target, which is fixed with a new remote protocol command to set this field in the shadow DP.

The third issue we hit with this was that the error checking and handling code in the ADI SWD logic was not detecting a failed read, which is now solved.

Please give `fix/stm32wl55-discovery` a try and let us know how you get on with all these fixes and improvements applied. With using connect-under-rest (`monitor connect_rst enable`) we have been able to get a clean scan of the target and attach to each core (though only one in each scan because as soon as the target is allowed to resume execution on the M4 core, debug dies instantly)

AJ528, 2025-01-08:
Hi Dragonmux,

Thanks so much for working on this issue! My debug cable from Samtec never arrived, and then I honestly got distracted. But I am happy to test and verify code functionality. My first impressions are that it is a huge improvement! I can connect, scan, debug my target, and set breakpoints.

```
(gdb) target extended-remote /dev/ttyACM0
Remote debugging using /dev/ttyACM0
(gdb) monit version
Black Magic Probe v2.0.0-rc1-9-g66eb6cc5, Hardware Version 6
Copyright (C) 2010-2024 Black Magic Debug Project
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>

(gdb) monit swd_scan
Target voltage: 3.3V
Available Targets:
No. Att Driver
 1      STM32WLxx M4
 2      STM32WLxx M0+
(gdb) monit jtag_scan
Target voltage: 3.3V
Available Targets:
No. Att Driver
 1      STM32WLxx M4
 2      STM32WLxx M0+
(gdb) attach 1
Attaching to Remote target
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
0x08000760 in ?? ()
(gdb) file ./bin/mcli.elf 
A program is being debugged already.
Are you sure you want to change the file? (y or n) y
Reading symbols from ./bin/mcli.elf...
(gdb) load
Loading section .isr_vector, size 0x138 lma 0x8000000
Loading section .text, size 0x1444 lma 0x8000138
Loading section .rodata, size 0xfc lma 0x800157c
Loading section .ARM, size 0x8 lma 0x8001678
Loading section .data, size 0x14 lma 0x8001680
Start address 0x0800152c, load size 5780
Transfer rate: 20 KB/sec, 578 bytes/write.
(gdb) b main
Breakpoint 1 at 0x800059e: file src/main.c, line 20.
Note: automatically using hardware breakpoints for read-only addresses.
(gdb) c
Continuing.

Breakpoint 1, main () at src/main.c:20
20	  sysclk_init();
(gdb) s
sysclk_init () at src/main.c:51
51	  SystemCoreClockUpdate();
(gdb) l main.c:20
15	static void Error_Handler(void);
16	
17	
18	int main(void)
19	{
20	  sysclk_init();
21	  UART_init();
22	
23	
24	  println_("hello world!");
(gdb) b main.c:24
Breakpoint 3 at 0x80005a6: file src/main.c, line 24.
(gdb) c
Continuing.

Breakpoint 3, main () at src/main.c:24
24	  println_("hello world!");
(gdb) q
```

I also tried enabling the `connect_rst` option, and I can't say I noticed any difference. What is the advantage to connecting to the target under reset vs not under reset? Here is my gdb log from that time:
```
(gdb) target extended-remote /dev/ttyACM0
Remote debugging using /dev/ttyACM0
(gdb) monitor connect_rst enable
Assert nRST during connect: enabled
(gdb) monitor version
Black Magic Probe v2.0.0-rc1-9-g66eb6cc5, Hardware Version 6
Copyright (C) 2010-2024 Black Magic Debug Project
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>

(gdb) monitor jtag_scan
Target voltage: 3.3V
Available Targets:
No. Att Driver
 1      STM32WLxx M4
 2      STM32WLxx M0+
(gdb) monitor swd_scan
Target voltage: 3.3V
Available Targets:
No. Att Driver
 1      STM32WLxx M4
 2      STM32WLxx M0+
(gdb) attach 1
Attaching to Remote target
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
0x080004ec in ?? ()
(gdb) file ./bin/mcli.elf 
A program is being debugged already.
Are you sure you want to change the file? (y or n) y
Reading symbols from ./bin/mcli.elf...
(gdb) b main
Breakpoint 1 at 0x800059e: file src/main.c, line 20.
Note: automatically using hardware breakpoints for read-only addresses.
(gdb) c
Continuing.
^C
Program received signal SIGINT, Interrupt.
0x0800076a in bufIsEmpty (buf=0x20000000 <rxBuffer>) at src/mcli.c:64
64	}
(gdb) run
The program being debugged has been started already.
Start it from the beginning? (y or n) y
Starting program: /home/alex/code/mcli/bin/mcli.elf 

Breakpoint 1, main () at src/main.c:20
20	  sysclk_init();
(gdb) monitor jtag_scan
Target voltage: 3.3V
You are now detached from the previous target.
Available Targets:
No. Att Driver
 1      STM32WLxx M4
 2      STM32WLxx M0+
(gdb) attach 1
A program is being debugged already.  Kill it? (y or n) y
Attaching to program: /home/alex/code/mcli/bin/mcli.elf, Remote target
0x080007d0 in cli_process () at src/mcli.c:80
80	}
(gdb) run
The program being debugged has been started already.
Start it from the beginning? (y or n) y
Starting program: /home/alex/code/mcli/bin/mcli.elf 

Breakpoint 1, main () at src/main.c:20
20	  sysclk_init();
(gdb) s
sysclk_init () at src/main.c:51
51	  SystemCoreClockUpdate();
(gdb) b main.c:24
Breakpoint 2 at 0x80005a6: file src/main.c, line 24.
(gdb) c
Continuing.

Breakpoint 2, main () at src/main.c:24
24	  println_("hello world!");
(gdb) q
A debugging session is active.

	Inferior 1 [Remote target] will be killed.

Quit anyway? (y or n) y
```

I can't promise it was a super-rigorous test, but everything seems to be working to me. Are there any other tests you'd like me to do to verify functionality?

AJ528, 2025-01-08:
> The primary difference here is that the target is held in reset from the
> moment scan starts (mon jtag or mon swd) to the moment you first attach
> to a core. Within your debugging setup, it won't make a difference. We
> merely noted it as a necessary step we had to take with whatever default
> firmware ST supplied on the STM32WL55 attached to this Nucleo board we've
> been using to debug with. Frustrating, but it has helped find some
> Interesting bugs.
> Everything else looks good though, so we'll mark this issue fixed when we
> open the PR from that branch so the two will close together. Thank you for
> the fast turn-around on testing this!

Of course! Thank you for taking the time to fix this issue. I appreciate it.

closed 2025-01-08 by esden