blackmagic-debug/blackmagic #2034: STM32WLxx: Start and Run commands never return

AJ528, 2024-12-22:
STM32WLxx: Start and Run commands never return

Hello,

I am working on programming an STM32WL55CC using a black magic debug probe. I have been able to connect to the MCU using the SWD interface and then load and step through code using gdb-multiarch. However, whenever I try to restart the code using `start` or `run`, the command never returns; it just hangs until I use ctrl+C to interrupt it. I would expect the code to restart and break on the entry to `main()`, but this doesn't happen. The only way I can restart my code and run from the beginning is using the `load` command.

I am using a black magic debug probe v1.10.2 and gdb-multiarch 12.1.

Here is a log showing everything working correctly until I try to restart using `start`:

```
GNU gdb (Ubuntu 12.1-0ubuntu1~22.04.2) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb) target extended-remote /dev/ttyACM0
Remote debugging using /dev/ttyACM0
(gdb) file bin/output.elf 
Reading symbols from bin/output.elf...
(gdb) monitor version
Black Magic Probe v1.10.2, Hardware Version 6
Copyright (C) 2010-2023 Black Magic Debug Project
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>

(gdb) monitor swd_scan
Target voltage: 3.3V
Available Targets:
No. Att Driver
 1      STM32WLxx M4
 2      STM32WLxx M0+
(gdb) attach 1
Attaching to program: /home/alex/code/remote_base_station/bin/output.elf, Remote target
0x1fff2596 in ?? ()
(gdb) break main
Breakpoint 1 at 0x800017c: file src/main.c, line 24.
Note: automatically using hardware breakpoints for read-only addresses.
(gdb) load
Loading section .isr_vector, size 0x138 lma 0x8000000
Loading section .text, size 0x1fd8 lma 0x8000138
Loading section .rodata, size 0x1dc lma 0x8002110
Loading section .data, size 0x4 lma 0x80022ec
Start address 0x080020c0, load size 8944
Transfer rate: 27 KB/sec, 745 bytes/write.
(gdb) s
Reset_Handler () at src/startup_stm32wl55xx_cm4.s:59
59	  mov   sp, r0          /* set stack pointer */
(gdb) 
Reset_Handler () at src/startup_stm32wl55xx_cm4.s:62
62	  bl  SystemInit
(gdb) 
SystemInit () at src/system_stm32wlxx.c:224
224	}
(gdb) c
Continuing.

Breakpoint 1, main () at src/main.c:24
24	  subghz_init();
(gdb) s
subghz_init () at src/subghz.c:40
40		LL_APB3_GRP1_EnableClock(LL_APB3_GRP1_PERIPH_SUBGHZSPI);
(gdb) start
The program being debugged has been started already.
Start it from the beginning? (y or n) y
Temporary breakpoint 2 at 0x800017c: file src/main.c, line 24.
Starting program: /home/alex/code/remote_base_station/bin/output.elf 
^C
Program received signal SIGINT, Interrupt.
0x1fff2ba6 in ?? ()
(gdb)
```

dragonmux, 2024-12-22:
The reason for this behaviour is not to do with BMD - your MCU is stuck executing its boot ROM (hence the 0x1fffxxxx address on Ctrl + C) and so the breakpoint being requested is never being hit. This is likely an issue with the state of the STM32's BOOT0 pin causing it to execute the ROM bootloader in preference for your firmware.

AJ528, 2024-12-22:
Thanks for the quick reply and pointing out my MCU is in boot ROM. That information allowed me to figure out the issue.

Like many STM32s, this MCU has an empty check mechanism that indicates if the first address in flash memory (0x08000000) is erased (0xFFFFFFFF). If it is, the MCU enters the system bootloader on reset. Critically, this check is usually only performed on a **Power Reset**. This is different from a System Reset. A System Reset can be triggered by toggling the reset pin, a watchdog event, a software reset, and other events. A Power Reset occurs when power is removed from the MCU.

Because the empty check only updates on a Power Reset, if you program an empty MCU using GDB's `load` command and immediately start debugging it, everything will appear to work because GDB sets the PC to the start of your program. However, the empty flag will remain set until MCU power is cycled. So if a System Reset is triggered (e.g. by running `start` or `run` in GDB), the MCU will jump to the system bootloader because it thinks the flash is still empty. Power-cycling the MCU will fix the issue at this point.

Here is the relevant paragraph from my MCU's reference manual:

> This empty check flag is updated only during the loading of option bytes: it is set when the content of the address 0x0800 0000 is read as 0xFFFF FFFF, otherwise it is cleared. A power reset or setting of OBL_LAUNCH bit in FLASH_CR is needed to clear this flag after programming of a virgin device to execute user code after a system reset. The EMPTY bit can also directly be written by software.

From section 4.3.2 of RM0453 Rev 5

closed 2024-12-22 by AJ528