blackmagic-debug/blackmagic #2046: firmware not working for BMP21 probe.

abdosalem490, 2025-01-08:
I have replicated [https://black-magic.org/_downloads/9268bd95a59bf58b68008004df51c31f/bmpm_v2_1c_schematic.pdf](BMP21 schematics) and tried to download the native firmware into it using segger j-link, but it doesn't work. I tried to debug it using segger j-link from the command line, it seems like stm32f103Cb has entered hard fault handler. I tried to build the source code for the BMP21 (I know it's a legacy board) but all the targets I see in meson build are this: 
 
[image]

can anyone help ?

dragonmux, 2025-01-08:
👋🏼 First off, a point of nomenclature.. what you're wanting to work with are not targets - those, strictly, are the things the firmware supports debugging. You are wanting to build for a specific probe. In your case, `native`.

Second, without knowing what exactly you tried to load, where, and with what option bytes set.. there's too little information here to be able to help.

You will need to load the project bootloader onto the STM32F103 first, and make sure it enumerates and boots. You can then load the firmware, the firmware does not include the bootloader in the image. You will need to have set the option bytes appropriately for the HW3 platform (v2.1c hardware is hardware ABI version 3) so the firmware can correctly determine what it's running on.

abdosalem490, 2025-01-08:
@dragonmux 
I am sorry I didn't elaborate on what I am trying to achieve.
I have made the to-be called debugger board for my needs and here is an image of it in the PCB design software:
[image]

after I received the board, I soldered all the components onto it and made sure that the MCU USB port is working by writing my own code for it. what I am trying to accomplish now it have the black magic firmware running on it so that I can use this board as a debugger for other boards.

I download the binary file for [Stable Release v1.10.2](https://github.com/blackmagic-debug/blackmagic/releases/tag/v1.10.2) and tried to download the binary file into my debugger board using a j-link but it wasn't working (doesn't appear under the ports tab in the device manager). I tried older release which is [Stable Release V1.6.1](https://github.com/blackmagic-debug/blackmagic/releases/tag/v1.6.1) but also, it's not working. 
I tried to debug the firmware but it seems like the binary I am trying to download isn't build for stm32f103cb because it seems like it's trapped inside a hardfault handler function.

can you tell what I am doing wrong?

dragonmux, 2025-01-08:
Everything in our original reply still applies. You still need to provision the bootloader, and option bytes. The firmware on its own is not sufficient for a from-scratch bringup. It's not about the release chosen, the board simply doesn't have the bootloader or option bytes to start with. We would also strongly advise using `main` (v2.0.0-rc1) and not v1.10.2 but that doesn't change our advise above.

Bootloader first, make sure that enumerates.

abdosalem490, 2025-01-08:
@dragonmux 
I am sorry but I am just a noob, I can't find the boot-bin directory, I have seen it mentioned on the official website [https://black-magic.org/knowledge/compiling-windows.html](Compiling on Windows)
[image]
I have searched in the downloaded repository and after running the command: 
`meson setup build --cross-file cross-file/native.ini`
and another time after running the command:
`meson compile -C build`
I got the binary and executable files for the target but no 'blackmagic_native_bootloader.elf' is found.

dragonmux, 2025-01-08:
It's not a directory, it's a specific target in the Meson build system - hence telling you to use `ninja` directly instead of `meson compile`. `ninja -C build` tells Ninja to operate in the `build` directory, what comes after that is the name of a build target [Ninja nomenclature, not project].

`ninja -C build boot-bin` will make you the bootloader .elf and .bin. Again, this is not a directory or source file, it is the name of a build target.