blackmagic-debug/blackmagic #2058: RTT stops working for nrf52840 since a7090a17

buganini, 2025-01-19:
I am using `Raytac MDBT50Q – 1MV2 (nrf52840)` + BMP, with SWDIO/SWDCLK, no nRST

RTT works before a7090a17, since that commit there is no output to the `xxxxxxxx3` interface.

```
[NG] 1d7c4ccc target_flash: Added a diagnostic for what is being erased when TARGET-level diagnostics are enabled
...
[NG] 9ac66f55 stlink: add config for the ISOL variant
...
[NG] 6cc8286e ctxlink/wifi_server: Move definitions and variables to file start
...
[NG] bed8cea7 ctxlink/platform: Move local variables to limit scope Also fix local variable "count" to match sscanf return type.
...
[NG] def92c78 remote: provide correct 3-byte buffer to `hexify()`
...
[NG] c3cdfbf2 ch32f1: Fixed a whole lot of clang-tidy lints, const-correctness and type issues
...
[NG] 260eea3b riscv_debug: Enabled non-halting memory I/O on targets that use System Bus for memory access
[NG] a7090a17 target: Rebuilt the non-halting I/O heuristic, replacing it with a target option
[OK] e44d0012 cortexm: Fix the float unit TOPT which was wrongly marked as for ARMv7-M only and accidentally conflcited with the ARMv8-M profile TOPT
...
[OK] f56c9c80 target: Switched the name for the RISC-V and its acceleration enable macros to `CONFIG_RISCV` and `CONFIG_RISCV_ACCEL`
...
[OK] e7957530 meson: Disabled some of the default targets to make the code fit again.
...
[OK] df49eb45 hosted/bmp_serial: Finished implementing the string logic needed to extract the type and version information from the product string to complete Windows device discovery
...
[OK] 7b11f503 ctxlink: Enabled switchable SWO on the platform
...
[OK] cb089bf3 rtt: demystify `fast_search()`
...
[OK] a85b6017 lpc43xx: fix part number typo for 433x and 435x
```

ALTracer, 2025-01-19:
Yes, this behaviour change is intended as PR1962 describes (to which your bisected first-NG commit hash points). I've tried to link PR2037 workaround to it, is it not discoverable enough? Now BMP halts any processor, including nRF52, as soon as it receives `mon rtt enable` command, to search entire SRAM (long pause) for control block and repeatedly poll ringbuffers (3 milliseconds order of magnitude pauses), which may trigger asserts or other timeouts in SoftDevice.
Please rebuild BMF with this patch yourself and report back whether it solves this on nRF52 family, i.e. returns valid log data not garbage. I cannot test on hardware I don't have, so I won't add it to 2037 for automatic CI binaries for you to download and flash.
```diff
diff --git a/src/target/nrf51.c b/src/target/nrf51.c
index 7400254f..b95e0f08 100644
--- a/src/target/nrf51.c
+++ b/src/target/nrf51.c
@@ -144,6 +144,8 @@ bool nrf51_probe(target_s *t)
 		uint32_t ram_size = target_mem32_read32(t, NRF52_INFO_RAM);
 		t->driver = "nRF52";
 		t->target_options |= TOPT_INHIBIT_NRST;
+		/* On nRF52 SoC, Cortex-M4F allows SRAM access without halting */
+		t->target_options |= TOPT_NON_HALTING_MEM_IO;
 		target_add_ram32(t, 0x20000000U, ram_size * 1024U);
 		nrf51_add_flash(t, 0, page_size * code_size, page_size);
 		nrf51_add_flash(t, NRF51_UICR, page_size, page_size);
```
CC @zyp

buganini, 2025-01-19:
Yes it works!

closed 2025-01-21 by buganini