blackmagic-debug/blackmagic #2087: Discrepancy between clang-format and action linter

robots, 2025-03-03:
Hi,

When working on samx5x.c file i have tried to format it using clang-format. But this failed lint phase of the PR workflow.

Clang-format wanted this change:
```
[robot@robot-work blackmagic]$ python run-clang-format.py --style="file:.clang-format" ./src/target/samx5x.c
--- ./src/target/samx5x.c	(original)
+++ ./src/target/samx5x.c	(reformatted)
@@ -141,7 +141,7 @@
 /* Non-Volatile Memory Calibration and Auxiliary Registers */
 #define SAMX5X_NVM_USER_PAGE   UINT32_C(0x00804000)
 #define SAMX5X_NVM_CALIBRATION UINT32_C(0x00800000)
-#define SAMX5X_NVM_SERIAL(n)   (UINT32_C(0x0080600c) + ((n) == 0 ? 0x1f0U : (n)*4U))
+#define SAMX5X_NVM_SERIAL(n)   (UINT32_C(0x0080600c) + ((n) == 0 ? 0x1f0U : (n) * 4U))

 #define SAMX5X_USER_PAGE_OFFSET_LOCK     0x08U
 #define SAMX5X_USER_PAGE_OFFSET_BOOTPROT 0x03U
```

robots, 2025-03-03:
Here is link to the failed action: https://github.com/blackmagic-debug/blackmagic/actions/runs/13634707862/job/38110473888

dragonmux, 2025-03-03:
This is a known problem with using newer clang-format with this code base. We intend to upgrade the clang-format used by the lint pass to clang-format-18 to grab a pile of fixes (such as this one) that were made in the newer release but didn't want to do that for v2.0 as it is a disruptive change (requires reformatting a pile of files).

This is entirely due to older clang-format being a little buggy around such macros.

dragonmux, 2025-03-03:
We'll note that whatever we do in relation to this, we're kinda damned either way: people using older clang-format such as comes with Debian stable or Ubuntu get the "right" behaviour (lint-passing) and don't complain, and people using newer get the "wrong" behaviour and complain; people using newer clang-format such as comes with Arch or the Clang APT repos get the right behaviour and don't complain, and people using the older get the wrong behaviour and do complain.

😅 So we get to choose which way we're going to have some subset of users complain - going with the newer clang-format though means getting a pile of bugfixes and it working better with our own setup, so this is why the path we're going to tread for v2.1.

robots, 2025-03-03:
I just wanted to point this out because it kicked me, while trying to clean the file. And yes, i am on bleeding edge - Arch.