blackmagic-debug/blackmagic #2107: BMDA does not work with CMSIS-DAP probes that have trace endpoint

bugadani, 2025-04-17:
Due to [this line](https://github.com/blackmagic-debug/blackmagic/blob/221c3031f57d284955836110c4db7777e954a0a7/src/platforms/hosted/bmp_libusb.c#L477-L478) CMSIS-DAP V2 probes with 3 endpoints are not recognised as `CMSIS_TYPE_BULK`.

This configuration is permitted by the [CMSIS-DAP documentation](https://arm-software.github.io/CMSIS-DAP/latest/dap_firmware.html#dap_bulk_usb):

> CMSIS-DAP v2 uses USB bulk endpoints. Optionally, support for streaming SWO trace is provided via an additional USB endpoint.

dragonmux, 2025-04-17:
Does changing the check from `== 2` to `>= 2` (or a check for either 2 or 3, more correctly) solve that? If so, please PR a fix and we'll get it merged for v2.0 as this is the kind of bug the release candidates are intended to help find and solve.

bugadani, 2025-04-17:
I don't really know, I don't have a C build environment set up. But just changing the comparison would mean that the trace endpoint would overwrite the data IN endpoint so the probe may be detected, but it would likely not work at all.

dragonmux, 2025-04-17:
Ah, okay - if you have the output for `lsusb` (or similar descriptor dumps) for a probe that actually has this optional 3rd endpoint, we can look at trying to build a fix then. The problem we currently have is that we don't have any eyes on what this 3 endpoint setup looks like as we have no adaptors that sport the optional 3rd endpoint. That's part of why this bug's gone undetected for a couple of years.

bugadani, 2025-04-17:
Unfortunately my j-link is CMSIS-DAP v1 only. As V2, I only have the one I'm currently building, which I don't think counts as a valid test case for now :) I can point at [one of the reference impls](https://github.com/ARM-software/CMSIS-DAP/blob/1fd47bed772ea40923472c90dfe11516e76033ee/Firmware/Examples/LPC-Link2/USBD_User_CustomClass_0.c#L350) that write EP2 to output SWO data.

The question is, how complete does blackmagic aim to be? If you don't plan to support reading SWO via bulk endpoint, then just ignoring the third endpoint should be sufficient for the time being. I don't expect probes to actually mix and match which endpoint is trace output, it would be horrendeous to try and identify them.

dragonmux, 2025-04-17:
It would be nice in the future to support dealing with the SWO endpoint.. but until that point in time, we'd like to be able to detect and handle adaptors with it correctly for debug all the same

closed 2025-04-17 by dragonmux

bugadani, 2025-04-17:
Ah nice, that change was just enough :)

dragonmux, 2025-04-17:
Perfect! Thank you very much for both finding this and figuring out a fix for it. It is greatly appreciated.

PR #2109 (2025-04-17): hosted/CMSIS-DAP: Do not discard devices with optional SWO endpoint

## Detailed description

This PR fixes the detection of CMSIS-DAP V2 adaptors which have the optional SWO trace USB bulk endpoint. This configuration is allowed by the [CMSIS-DAP documentation](https://arm-software.github.io/CMSIS-DAP/latest/dap_firmware.html#dap_bulk_usb). The documentation also prescribes the order of USB endpoints, meaning we are safe to ignore the optional endpoint.

## Your checklist for this pull request

* [x] I've read the [Code of Conduct](https://github.com/blackmagic-debug/blackmagic/blob/main/CODE [... cut]