blackmagic-debug/blackmagic #2119: STM32F411CEU - firmware not working

janholtzhausen, 2025-05-14:
Followed firmware building instructions, flashed via dfutil as well as stm32 programmer, every combination of bootloader at 0x08000000 and bin at 0x08004000, bin at 0x08000000, compiled on Ubuntu dev box, as well as on MSYS2 on windows 11.
Result:
USB\DEVICE_DESCRIPTOR_FAILURE
or no device at all.

Are we SURE this is supposed to work on a black pill???

dragonmux, 2025-05-14:
Please can you share which version of the firmware you're trying to build, with what options, compiler, and how you're configuring and running the build?

The details provided so far leave too much to guess-work to properly help you beyond providing guidance.

ALTracer, 2025-05-14:
Please demonstrate your ordered board contains 25MHz HSE crystal, and not newer 8MHz one which is not yet supported by platform init code. Also you can use some other debugger (or even WeAct HID CMSIS-DAP) to perform inception debugging yourself and troubleshoot it. It works as recently as v2.0.0-rc2 and should not be broken.
P.S. DfuSe MaskROM autodetects 8/24/25/26 integer MHz crystals via internal timer loopback, I'm thinking of doing the same if more users start flashing 8MHz boards.

ALTracer, 2025-05-14:
> 8Mhz crystal

Please patch locally the rcc config line and rebuild both binaries, this basically changes the PLL input prescaler from /25 to /8, the rest of multipliers are fine.

https://github.com/blackmagic-debug/blackmagic/blob/069fa4260f85118dbd1e56c7900f5d9d37a44a32/src/platforms/common/blackpill-f4/blackpill-f4.c#L95

Multiplier enums come from 3 board variants.
HSE25 and HSE8 are already supported by libopencm3_stm32f4.a, no need to reinvent that.

Otherwise the board may get stuck waiting for PLL to lock onto signal too slow, and/or USB48ck will be out of spec. HSI8 is not suitable for USBOTG_FS, and there is no CRS autotrim on F4x1Cx.

closed 2025-05-14 by janholtzhausen