blackmagic-debug/blackmagic #2124: Single step exiting an interrupt causes it to behave as if 'continue' was typed

kairoswater-jason, 2025-06-04:
This same issue is present in many other hardware debuggers - including ST's gdb-server + debug probe firmware. (But not in probe-rs, which has its own issues)

On a STM32G0B1
using a STLink V2.1 
With memory protection and privileged/thread mode

[image]

When using the `stepi` instruction to exit an ISR causes it as if 'continue' was typed ("zooms back around to the entry breakpoint")

kairoswater-jason, 2025-06-04:
One of the gdb people linked me to [1]

[1] https://sourceware.org/bugzilla/show_bug.cgi?id=28440

dragonmux, 2025-06-04:
With thanks to that hint, we've figured out that this was down to not saying we supported signaled stepping in response to `vCont?` - with that addressed as in the linked PR, you're able to step the way you were expecting. We suspect something is still off with the thread info part still, but that can be addressed seperately.

closed 2025-06-13 by esden