blackmagic-debug/blackmagic #2146: Build blackmagic app to run on arm orangepi

pistoletov1974, 2025-08-30:
Hi! I want to try running a debugging application on a single-board computer, but I get an error when I try to build it.
`root@orangepipc:~/blackmagic# meson setup build -Dprobe=''
The Meson build system
Version: 0.53.2
Source dir: /root/blackmagic
Build dir: /root/blackmagic/build
Build type: native build

meson.build:196:3: ERROR: Expecting rparen got string.
                f'@bmf_base_name@_firmware',
   ^
For a block that started at 195,21
        bmf_elf = executable(
                     ^
`

 Does anyone have experience running an application on a device like this? This would be a great way to debug remote devices.

dragonmux, 2025-08-30:
For that setup, you want to build BMDA - drop the `-Dprobe=''` part. `meson setup build` w/ the gpiod dev libraries installed will get you a suitable binary to work with your SBC.

pistoletov1974, 2025-08-31:
> For that setup, you want to build BMDA - drop the `-Dprobe=''` part. `meson setup build` w/ the gpiod dev libraries installed will get you a suitable binary to work with your SBC.
I installed  gpiod lib - apt install gpiod
and try to build BMDA without Dprobe option. Result is the same error.
May be problem in meson version?

```
root@orangepipc:~/blackmagic# meson setup build
The Meson build system
Version: 0.53.2
Source dir: /root/blackmagic
Build dir: /root/blackmagic/build
Build type: native build

meson.build:196:3: ERROR: Expecting rparen got string.
                f'@bmf_base_name@_firmware',
   ^
For a block that started at 195,21
        bmf_elf = executable(
                     ^

A full log can be found at /root/blackmagic/build/meson-logs/meson-log.txt
```